gnulabis/UTFT-ESP #1: Drawbitmap ESP8266

danbicks, 2016-01-31:
Hi, first of all fantastic job, excellent port of UTFT.

I have a couple of issues. The first one is trying to draw a bitmap using the example code for AVR on ESP does not work. I think this may be something silly, I tried the info icon but seem to get an error then with the drawbitmap routine.

The other issue is the speed for clearing the display seems a little slow in comparison to the Adafruit library. I have made sure that the hardware speed for SPI is at 40mhz, this does not make a difference. Is there any way of speeding this up. Otherwise fantastic work.

I am using 1.6.5 IDE with 2.0.0 esp build.

Thanks

Dans

gnulabis, 2016-01-31:
Hey there,

thanks for the feedback, much appreciated. I'm currently moving to a new apartment and I have very limited access to my equipment, but I'll have a look at it as soon as I set everything up in my new house.

Cheers

gnulabis, 2016-03-02:
Hi again,

First of all, sorry for the delayed answer.

If you are still interested in this, I had a look at it and I think I can help.

Also, let's leave the "slow clear display" aside for the moment.

So, examples from the original UTFT are not supposed to work anyway. Only the ESP8266-specific examples work. Nevertheless, I tried compiling the original UTFT_Bitmap demo for AVR and I managed to do so, following these basic steps:

1. Replace `#include <avr/pgmspace.h>` with `#include<pgmspace.h>` in **all** the sketch files
2. Rename file `info.c` to `infobmap.c` (there is a conflict with the name "info")
3. In `infobmap.c`, replace `const unsigned short info[0x400] PROGMEM` with `const unsigned short infobmap[0x400] PROGMEM`
4. In the main sketch file, replace all mentions of `info` with `infobmap`
5. Make sure you change the pin numbers/names in the UTFT constructor as well
6. If using IDE 1.6.5, `#include <SPI.h>`

I'm not sure if it will work, but it should at least compile like this. Chances are that you have to add a couple of yield() statements here and there, much like I did the the ESP8266-specific examples.

If you run into any errors, please copy them verbatim to your report, it helps.

If you manage to make it work, please consider forking the repository, creating an ESP8266-specific UTFT_Bitmap example and issuing a pull request. I'll happily add it to the current list of examples.

gnulabis, 2016-09-13:
Closing issue due to lack of feedback

closed 2016-09-13 by gnulabis