gnulabis/UTFT-ESP #2: Problem with Arduino IDE 1.6.5

lukbe, 2016-02-23:
Hello, i have problem with arduino IDE 1.6.5 and esp8266 platform package for Arduino version 2.0.0.
If i compile project i receive this error message:

```
Arduino: 1.6.5 (Windows 8.1), Vývojová deska: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

UTFT_Demo_320x240_Serial_SW.ino: In function 'void loop()':
UTFT_Demo_320x240_Serial_SW.ino:53:73: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
UTFT_Demo_320x240_Serial_SW.ino:56:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
UTFT_Demo_320x240_Serial_SW.ino:75:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
UTFT_Demo_320x240_Serial_SW.ino:83:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
UTFT_Demo_320x240_Serial_SW.ino:91:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
UTFT_Demo_320x240_Serial_SW.ino:313:45: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
UTFT_Demo_320x240_Serial_SW.ino:314:51: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
UTFT_Demo_320x240_Serial_SW.ino:315:50: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
UTFT_Demo_320x240_Serial_SW.ino:319:52: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
In file included from C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\UTFT.cpp:88:0:
C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\hardware/esp8266/HW_ESP8266.h:1:17: fatal error: SPI.h: No such file or directory
 #include <SPI.h>
                 ^
compilation terminated.
```

If i use arduino IDE version 1.6.7 i not receive :

```
+1: In file included from C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\UTFT.cpp:88:0:
C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\hardware/esp8266/HW_ESP8266.h:1:17: fatal error: SPI.h: No such file or directory
 #include <SPI.h>
                 ^
compilation terminated.
```

I receive only this warning:

```
C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\examples\ESP8266\UTFT_Demo_320x240_Serial_SW\UTFT_Demo_320x240_Serial_SW.ino: In function 'void loop()':

C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\examples\ESP8266\UTFT_Demo_320x240_Serial_SW\UTFT_Demo_320x240_Serial_SW.ino:53:73: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

     myGLCD.print ( "* Universal Color TFT Display Library *", CENTER, 1 );

                                                                         ^

C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\examples\ESP8266\UTFT_Demo_320x240_Serial_SW\UTFT_Demo_320x240_Serial_SW.ino:56:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

     myGLCD.print ( "<http://www.RinkyDinkElectronics.com/>", CENTER, 227 );

                                                                          ^

C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\examples\ESP8266\UTFT_Demo_320x240_Serial_SW\UTFT_Demo_320x240_Serial_SW.ino:75:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

     myGLCD.print ( "Sin", 5, 15 );

                                 ^

C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\examples\ESP8266\UTFT_Demo_320x240_Serial_SW\UTFT_Demo_320x240_Serial_SW.ino:83:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

     myGLCD.print ( "Cos", 5, 27 );

                                 ^

C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\examples\ESP8266\UTFT_Demo_320x240_Serial_SW\UTFT_Demo_320x240_Serial_SW.ino:91:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

     myGLCD.print ( "Tan", 5, 39 );

                                 ^

C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\examples\ESP8266\UTFT_Demo_320x240_Serial_SW\UTFT_Demo_320x240_Serial_SW.ino:313:45: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

     myGLCD.print ( "That's it!", CENTER, 93 );

                                             ^

C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\examples\ESP8266\UTFT_Demo_320x240_Serial_SW\UTFT_Demo_320x240_Serial_SW.ino:314:51: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

     myGLCD.print ( "Restarting in a", CENTER, 119 );

                                                   ^

C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\examples\ESP8266\UTFT_Demo_320x240_Serial_SW\UTFT_Demo_320x240_Serial_SW.ino:315:50: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

     myGLCD.print ( "few seconds...", CENTER, 132 );

                                                  ^

C:\Users\lukbe\OneDrive\Dokumenty\Arduino\libraries\UTFT\examples\ESP8266\UTFT_Demo_320x240_Serial_SW\UTFT_Demo_320x240_Serial_SW.ino:319:52: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

     myGLCD.print ( "Runtime: (msecs)", CENTER, 210 );

                                                    ^
```

WHY ?

gnulabis, 2016-03-02:
So, there are two things reported in your issue. I'll address them separately.

### Issue 1: String conversion warnings
To get rid of the warnings, you just need to convert all the constant char pointers to String. I've updated all UTFT examples to do so. Check out the updated examples in commit 60c840088062759b41e07c35192651d4b6b81eb7

### Issue 2: missing SPI.h in Arduino IDE 1.6.5
You are absolutely right, I had seen this behaviour as well and then I forgot about it when I switched to 1.6.6.

This is a quirk of the Arduino IDE itself with respect to 3rd party cores (such as the ESP8266). It has been "fixed" in version 1.6.6 and later of the IDE. You can read more about it at arduino/Arduino#2573.

For Arduino IDE 1.6.5, the workaround is to also include SPI.h in your own sketch.

I've updated all UTFT examples to do so. Check out the updated examples in commit d638c604a490063d6adbedf06324c0b3b361ce36

I've tested this with 1.6.5, 1.6.6 and 1.6.7, it seems to be ok now.

closed 2016-03-02 by gnulabis

lukbe, 2016-03-02:
Ahoj thank you for answer. Please can you send me how i can connect display
with ESP ? Thank you

gnulabis, 2016-03-02:
Sorry, but I cannot help you with that. It really depends on your setup.

lukbe, 2016-03-02:
( ILI9341_S5P, 15, 5, 4 );

15=CS
5= RST
4 = ???

And as i have found: SCK and MOSI (GPIO14 and GPIO13)

And D/C ???

Help me please ....

lukbe, 2016-03-02:
i have connect it:

/*
 LCD --->> NodeMCU
  Vcc--->>Vcc
  GND--->>GND
  CS--->> D8  (1k ohm odpor)
  Reset--->>D1
  D/C--->> D2 (1k ohm odpor)
  SDI(mosi)--->>D7  (1k ohm odpor)
  SCK--->>D5  (1k ohm odpor)
  LED--->>Vcc
*/

And it work fine :)