gnulabis/UTFT-ESP #3: pin clarification

ppostuma, 2016-04-14:
Hi - this is probably stupid basic, but what pin does the INT SER=0 utilize? Would the SD_MOSI and SD_SCK pins work for the TFT serial communication as well? (My board uses the S6D1121 controller and is supposed to be capable of serial LCD addressing, but the documentation is woefully inadequate.) Not sure how to connect, and I don't ruin my hardware by making bad connections.

gnulabis, 2016-04-14:
Did you check the [official UTFT list of supported controllers](https://github.com/gnulabis/UTFT-ESP8266/blob/master/UTFT/Documentation/UTFT_Supported_display_modules_%26_controllers.pdf)?

AFAIK, **S6D1121 is only supported by UTFT in parallel mode** (8 and/or 16 bit). Furthermore, UTFT-ESP8266 only supports serial mode (for now at least, although the limited number of free IO pins on the ESP8266 makes it hard to work with displays in parallel mode anyway).

See also the Project Summary section in the [README file](https://github.com/gnulabis/UTFT-ESP8266/blob/master/README.md).

ppostuma, 2016-04-15:
Yes, it is supported, but you're right, only in parallel mode. Sigh. I can add pins, but I'd much rather use hardware SPI. Gonna get me a new display. Appreciate the answer and your hard work!

closed 2016-04-15 by ppostuma