gnulabis/UTFT-ESP #5: Pinout descriptor

cdtsilva, 2016-07-20:
Can someone be kind enough to clarify the pinouts for this?

I'm trying to test the display (which works fine under arduino) without success.
I'm using pins 13 and 14 for MOSI and SCK respectivelly (HW)

The documentation states:
UTFT(byte model, int CS, int RST, int SER=0);

So that sets the CS, Reset and an unkinown SER = 0 -> What is this!??
How do I declare the argument for the D/C Pin?

Thanks

gnulabis, 2016-07-20:
SER is D/C.

The name comes from the original UTFT constructor for non-HW SPI:

https://github.com/gnulabis/UTFT-ESP8266/blob/master/UTFT/UTFT.cpp#L144

I'm not sure why the original author chose it, but I kept it the same to make it easier for people already familiar with UTFT.

cdtsilva, 2016-07-20:
Thanks. I assumed so, but still no luck. I Replaced the 2.2" display with a 2.8 and that worked fine.
Both displays work fine on arduino using a same code, could it be a initialization issue!?

gnulabis, 2016-07-20:
It's certainly not the same code on Arduino, because HW SPI is only supported on ESP8266. If you would use the same code on another platform, you would get an error about an undefined function.

Can you also try with the other constructor (the non-HW SPI)?

It's the one that you are probably also using on Arduino.

And a couple of questions:

1. Just to be sure, when you say that it works on Arduino, you are using UTFT there as well, right?

2. Which displays?

3. When you say that it doesn't work, could you be a bit more specific? No image at all, something else?

cdtsilva, 2016-07-20:
Sorry, let me clarify. What I meant is that I can interchange between the 2.2 and 2.8" displays using a given code. I tried both UTFT and Several variants of the Adafruit library: 

Using this library the 2.2 fails to initialize - It shows only a white background, as it would otherwise do when the back light is powered up.

This is the 2.2" display:
http://www.ebay.co.uk/itm/2-2-LCD-2-2-inch-SPI-TFT-LCD-Display-240x320-ILI9341-51-AVR-STM32-ARM-PIC-/301884935685?hash=item4649be8e05:g:CsEAAOSwzgRW0CeG

This is the 2.8" display
http://www.ebay.co.uk/itm/New-2-8-240x320-SPI-TFT-LCD-Touch-Panel-Serial-Port-Module-PCB-ILI9341-5V-3-3V-/351337012012?hash=item51cd515b2c:g:OVQAAOSwZjJU~rWj

gnulabis, 2016-07-20:
Despite your effort, I'm more confused now!

I now know of two mcus (Arduino, ESP8266), two displays (2.2 and 2.8) and at least two libraries (this port of UTFT, Adafruit). These lead to 8 possible combinations:

a) Arduino + 2.2 + UTFT
b) Arduino + 2.8 + UTFT
c) Arduino + 2.2 + Adafruit
d) Arduino + 2.8 + Adafruit
e) ESP8266 + 2.2 + UTFT
f) ESP8266 + 2.8 + UTFT
g) ESP8266 + 2.2 + Adafruit
h) ESP8266 + 2.8 + Adafruit

So, could you please indicate which of the above combinations you have already tried and, from those, which ones worked for you and which not?

Also, as I suggested in my previous reply, could you try with the non-HW SPI version as well?

UTFT(byte model, int RS, int WR, int CS, int RST, int SER);

Thanks

PS: in UTFT, RS is the data (MOSI), WR is the clock (SCK). The rest you know already from the HW SPI version.

cdtsilva, 2016-07-20:
I haven't been able to compile the adafruit library for the ILI9341 on the ESP8266. I get this error:

C:\Users\Carlos\Documents\Arduino\libraries\marekburiak-ILI9341_due-1c835c4/SystemFont5x7.h:33:26: fatal error: avr/pgmspace.h: No such file or directory

 #include <avr/pgmspace.h>

On a UNO it works fine with both the 2.2 and 2.8" displays, so I know the display is OK.
In regards to using HW and SW on the ESP8266, I tried both, same scenario. 2.8 works fine, 2.2 doesn't.

Thanks

gnulabis, 2016-07-22:
Sorry for the delay, busy times.

On the UNO, you have the 2.2 display also working with UTFT? If yes, are you using the original UTFT, or this port?

I cannot really make something out of this. As you can see from the commits in this repository, I did not modify any of the initialization and/or display support code. If it is a supported chipset (which it is) and it is working with Arduino+UTFT, it should also work here.

gnulabis, 2016-09-13:
Hi again,

I just got my hands on this 2.2 display:
https://www.amazon.fr/gp/product/B00HCZHHRU/

It even has the yellow plastic around the pin header like yours. It's probably the same display as the one you are having problems with.

Anyway, long story short, it worked out of the box for me. I tested it with Arduino IDE 1.6.6 and 1.6.11, as well as the esp8266 package versions 2.0.0 and 2.3.0. My esp8266 module is a NodeMCU v1.0.

I connected:
- VCC and LED pins to the 3V3 output of my esp8266
- ground to ground
- SCK to GPIO14 and SDI (MOSI) to GPIO13
- RST to GPIO15
- CS to GPIO5
- D/C to GPIO4

I left SDO (MISO) unconnected.

I tested it by running the ESP8266 UTFT_Demo_320x240_Serial examples, both HW and SW.

gnulabis, 2017-03-15:
Closing this due to inactivity

closed 2017-03-15 by gnulabis