gnulabis/UTFT-ESP #9: Cannot print floats

emogg, 2016-09-06:
What am I missing? This just prints a long line of junk on my tft. The "UTFT_Demo_320x240_Serial_HW" demo works perfectly but doesn't have a printNumF example.

    double testnum = 123.45;
    myGLCD.printNumF ( testnum, 1, CENTER, 150 );

Platformio
[env:nodemcu]
platform = espressif
framework = arduino
board = nodemcu

I downloaded UTFT lib from here, none available in platformio, except for "UTFT_SdRaw" which calls for UTFT!

gnulabis, 2016-09-07:
You are not missing anything, the library does :-)

function `_convert_float()` which is used internally by `UTFT:printNumF()` and is defined in source file [HW_ESP8266.h](https://github.com/gnulabis/UTFT-ESP8266/blob/master/UTFT/hardware/esp8266/HW_ESP8266.h#L94) is still empty.

I will fill it in (unless you feel inclined to do so!).

emogg, 2016-09-07:
Thank you, thank you. I have modified my copy of HW_ESP8266.h like so:

void UTFT::_convert_float ( char *buf, double num, int width, byte prec ) {
    dtostrf(num, width, prec, buf);
}

It works a treat now. Thank you very much.

closed 2016-09-07 by emogg

gnulabis, 2016-09-07:
Could you please commit it to your repository and do a pull request?

emogg, 2016-09-07:
On 07/09/16 11:21, gnulabis wrote:
> Could you please commit it to your repository and do a pull request?
>

I'd like to do that, but I'm far too much of a novice. Sorry I can't 
oblige at the moment.

gnulabis, 2016-09-07:
OK, no problem, I added the extra line. 575df9b3449a8ed12d3cd1858cbc779541df50a6