gnulabis/UTFT-ESP #11: No display updates on active serial terminal

Jorgen-VikingGod, 2016-11-13:
On my setup I have strange issues:
If the serial terminal is on - the TFT display do not update any content.
As far as I close the terminal and reset the ESP8266, it boots up as normal and the display lights up and shows up all the stuff I want.

**WeMos d1 mini with following wiring:**
```
// TFT display
// using HW-SPI on WeMos d1 mini
//      TFT_SCK     D5
//      TFT_MOSI    D7
//      TFT_MISO    D6 >> not connected
//      TFT_RST        >> connected to 3.3V
//      TFT_LED        >> connected to 3.3V
#define TFT_CS      D8
#define TFT_DC      D4
UTFT myGLCD(ILI9341_S5P, TFT_CS, -1, TFT_DC);
```

gnulabis, 2016-11-14:
Sounds like a conflict in pin mapping?

Did you have a look at https://github.com/esp8266/Arduino/blob/master/doc/reference.md and in particular the section about the serial interface?

Could it be that you have Serial0 swapped, using GPIO13 and GPIO15?

Does software SPI work? Using software SPI you could use different pins and see if (and where) you get the conflict.

Jorgen-VikingGod, 2016-11-14:
Sounds logic to me; but in my sketch I do not call `Serial.swap()` 
just on top of `setup()` method `Serial.begin();`
I will try Software SPI today - but this is no option for me, because of update speed reasons...

gnulabis, 2016-11-14:
I suggest SW SPI as a debugging tool, to see if there is an issue with the pins used for HW SPI, not as a final solution. I would start with exactly the same pin map used in HW SPI and then change pins one at a time.

Having said that, with the current implementation there is no significant improvement over SW SPI in speed when you use HW SPI. Of course, this will change when I find the time to review and factor in pull request #7...

gnulabis, 2016-11-15:
I tried yesterday your wiring with my NodeMCU and it worked without a problem, using Serial.begin() in setup() and a couple of Serial.println() which I was monitoring with the Arduino IDE serial monitor.

Jorgen-VikingGod, 2016-11-15:
I try to run it again and it works now...
Don't know why it did not work before. Maybe my wiring is not stable enough (=breadboard)
Thank you for your support.

closed 2016-11-15 by Jorgen-VikingGod