gnulabis/UTFT-ESP #12: Problems with the esp8266wifi library

kingw12jnsd, 2017-03-15:
Hello there! I have a problem very confused, there is a simple procedure, the code is as follows:
`#include <UTFT.h>`
`#include <ESP8266WiFi.h>`
`void setup () {}`
`void loop () {}`
The above code runs error, the error message is as follows:
C: \ users \ Administrator \ appdata \ local \ arduino15 \ packages \ esp8266 \ tools \ xtensa-lx106-elf-gcc \ 1.20.0-26-gb404fb9-2 \ xtensa-lx106-elf \ include \ c ++ \ 4.8.2 \ Bits \ move.h: 166: 28: error: macro "swap" requires 3 arguments, but only 2 given
     Swap (_Tp & __a, _Tp & __b)
Stl_pair.h: 194: 21: error: macro "swap" requires 3 arguments, but only 1 given
Stl_algobase.h: 147: 22: error: macro "swap" requires 3 arguments, but only 2 given
Allocator.h: 171: 21: error: macro "swap" requires 3 arguments, but only 2 given
I use the ESP-12, Arduino IDE 1.8.1, the two libraries are used alone without any problems, but put together will be wrong, what is the reason for this? How to fix?Looking forward to reply,thanks a lot!

gnulabis, 2017-03-15:
Hi,

I'll have a close look at it this evening. Until then:

1. I've never tried it yet with Arduino IDE 1.8.1, maybe that's the problem.
2. Can you compile, download and run the included examples?

kingw12jnsd, 2017-03-15:
Thanks for your reply!
If I use the UTFT library only, or just the ESP8266WiFi library, I can compile, download and run examples without any error, but when the two libraries included in the same program, the compiler will go wrong.

gnulabis, 2017-03-15:
Aaah, sorry, I previously misunderstood.

Quick googling reveals this:
https://github.com/esp8266/Arduino/issues/798

Long story short, can you try reversing the order of your includes? WiFi first, then UTFT.

kingw12jnsd, 2017-03-15:
Haaaaaaaa!

IT'S VERY THANKS FOR YOU!

I moved the ESP8266WiFi library to the front of the UTFT library, the problem be solved!

It is amazing! And THANKS again!

gnulabis, 2017-03-15:
No problem, good luck with your project

closed 2017-03-15 by gnulabis