gnulabis/UTFT-ESP #13: drawBitmap function shows corrupted image

danriches, 2017-03-16:
Hi,

Great library by the way, however after converting a small 37 x 54 png file using the image converter and adding it to my sketch folder the drawBitmap function draws the image twice next in a row and shows corrupted screen pixels underneath it. I'm using an ILI9341 with touchscreen and had to mod the flame.c file to use #include <pgmspace.h> rather than the avr one. Is there any fix for this or should I switch to the Adafruit library that has now been ported to the ESP8266? I've attached the original image for reference.

Thanks, Dan

[image]

gnulabis, 2017-03-16:
To be honest, I've never used the image converter tools. I don't even have a Windows OS installed anywhere to test them.

Would it be possible for you to send me the flame.c output of the tool? Preferably without any modifications, so that I can see how exactly it comes out of the tool.

Thanks!

danriches, 2017-03-29:
Hi,

I've attached it with a .txt extension. Thanks for any input on this.

Dan

[flame.c.txt](https://github.com/gnulabis/UTFT-ESP8266/files/878101/flame.c.txt)

studuino, 2017-08-10:
Hi,
I too am having problem with drawing bitmap. For given bitmap example in AVR/UTFT_bitmap, its fine (with filling up images with "for" loop). When drawing individual image, it shrink the sizes to about two folds and generate another duplicated image right beside the first one and gibberish at the bottom.

Thanks

gnulabis, 2017-08-10:
Ok, I fixed this, thanks to both of you for reporting this.

@danriches, apologies for taking so long to work on it, I hope you are still interested in the fix.

Please keep in mind that for this fix to work, you have to declare the images in your code as `extern unsigned short` (instead of `extern unsigned int`).

Please try out the latest commit in master and let me know so that I can close the issue.

Cheers

studuino, 2017-08-19:
@gnulabis Thanks for the fix. Works like a charm. Cheers.

closed 2017-08-21 by gnulabis