gnulabis/UTFT-ESP #14: Problem with ESP8266 (nodemcu) and ILI9481(3.2")

agardek, 2017-03-16:
I'm trying to test the display without success (only white screen).

I connected:
VCC and LED pins to the 3V3 output of my esp8266
gnd to gnd
SPI_CLK to GPIO14
SPI_MOSI to GPIO13
LCD_RST to GPIO15
LCD_CS(Chip selection) to GPIO5
LCD_RS(Command/Data selection) to GPIO4

//UTFT(byte model, int CS, int RST, int SER=0);
UTFT myGLCD ( ILI9481, 5, 15, 4 );

Below pinout form the display:
[image]
[image]

What I'm doing wrong?

gnulabis, 2017-03-16:
According to the original UTFT documentation ( UTFT-ESP8266/UTFT/Documentation/UTFT_Supported_display_modules_&_controllers.pdf), your screen is only supported in 16bit parallel mode, not serial. Sorry for that.

I'll see if I can make the library complain when such a screen is used with the esp8266.

Another option would be for the esp8266-utft to also support parallel interfaces, but I don't want to spend time on that, chances are that you don't even have 16+ pins on your esp8266 module to control the screen in 16-bit parallel mode, right?

agardek, 2017-03-17:
Thanks for answer.
I have only 10 pins GPIO.
In that case I'm going to buy another display - 2.8" with Serial Port Module and ILI9341 controller

closed 2017-03-17 by agardek

gnulabis, 2017-03-19:
Yep, that's why I haven't bothered so far with parallel interface displays for esp8266. It just doesn't have enough I/O available to drive them in a realistic application.

Good luck with your project.