gnulabis/UTFT-ESP #15: ESP32 port.

GabeHC, 2017-04-04:
Great job, Can you make it ESP32 compitable?

gnulabis, 2017-04-04:
Hi, thanks.

Not really, not now anyway, since I don't have the hardware to test it.

Cristianistrate, 2017-10-21:
Any news on ESP32 Port ?

gnulabis, 2017-10-24:
Hi, no, unfortunately I still don't have an ESP32.

Osmosis311, 2018-01-16:
I'm sure the community would be happy to send you one.  Adafruit has an ESP32 "Feather" board for only $20.

gnulabis, 2018-01-16:
Well, I had to order a new TFT module, so I ordered a NodeMCU ESP32 module as well.

Stay tuned...

gnulabis, 2018-01-27:
I just pushed 3ebc408c2d9cbe038315b996199eb85a2f7e7175, which adds preliminary support for ESP32. I briefly tested it with hardware SPI and it works. I'll merge it to master once I've done a few more tests.

commit 79f4efaf (2018-01-27): Update README.md to reflect support for ESP32. Closes #15.
closed 2018-01-27 by gnulabis