gnulabis/UTFT-ESP #17: SCK stays high after UTFT::drawline()

kenjones007, 2017-08-30:
When using a ESP8266 with a EastRising 1.8" Display using Software SPI (i modified your Driver a bit), i saw that my SCK line stayed high, after drawing a line, but only if the line is not vertical or horizontal. After checking your source, i think i found a litte bug:

UTFT_ESP8266.cpp
Line 844 + 866: you have to change **return** to **break**

a return will leave this function without setting sbi.

gnulabis, 2017-08-31:
Hi,

Looking at the code that you mention, I see indeed a potential problem, but it is with CS, not SCK.
Line 828 has: `cbi(P_CS, B_CS)`

So, I'm not sure that you have found the *right* bug for your problem.

Even so, this is something that is part of core UTFT, it has nothing to do with the ESP8266, it is common to all platforms and, as such, it should be reported upstream to the UTFT creator:
http://www.rinkydinkelectronics.com/contact.php

kenjones007, 2017-08-31:
Hi, yeah, you're right its CS but afterwards SCK stays high. I've contacted Henning Karlsen and he confirmed, that this should be "breaks" instead of "returns". But i have no feedback if he'll change that in his library. This problem occours only, if you draw diagonal lines. As soon as you draw an additional single H or V line, a dot or something else, it works. So i think this problem never really mattered for most users. Anyway, i fixed it in my copy of this library and since then it works flawless. Just wanted to inform you.  :+1: