gnulabis/UTFT-ESP #18: Working config for ILI9225 and Esp8266

dariocdj, 2018-02-14:
Hello, i'm trying your lib with an ili9225 and esp8266.
I've used UTFT myGLCD (ILI9225B, 15, 5, 4 ); and connected CS to gpio15, RST to gpio 5 and what is SER? i've put RS SPI Data command to 4.... but doesn't work.
Can you post a complete connection between ili9225 and esp8266 ? 
This is pinout : 
display			esp	
1 vcc			VCC
2 gnd			GND	
3 gnd			GND
4 nc
5 nc
6 led			VCC o pwm
7 clk spi clock		
8 sdi spi mosi		
9 RS SPI Data command
10 rst reset		
11 CS SPI Chip Select

Thanks
Dario

gnulabis, 2018-02-14:
Hi, check #5 for an explanation of SER (it's the command/data bit, in your case "RS").

Your wiring seems ok to me. Maybe double-check your SPI clk and data connections?

Also, ILI9225 support was added to this repository in 87c255f7395ebf5527fda1d284224ba097a20944 by @olegator77. Maybe you can check also with that user? I have no personal experience with this display.

dariocdj, 2018-02-14:
7 clk spi clock                    CONNECTED TO GPIO 14
8 sdi spi mosi                    CONNECTED TO GPIO 13
9 RS SPI Data command   CONNECTED TO GPIO4
10 rst reset                        CONNECTED TO GPIO5
11 CS SPI Chip Select        CONNECTED TO GPIO15

This is my connection and i'm trying to init with UTFT myGLCD (ILI9225B, 15, 5, 4 );

dariocdj, 2018-02-14:
Ok it works, sorry but my adapter has printed GPIO 4 and 5 inverted (WTF!). Thanks

closed 2018-02-14 by gnulabis

dariocdj, 2018-02-16:
Please how can i use this display if clk is also used by hx711 adc in the same node?