gnulabis/UTFT-ESP #19: 180 Rotate Display in Portrait Mode

dariocdj, 2018-02-20:
Hello, i need to rotate of 180° portrait mode of display (for pcb reason).

How i can modify this code to do this : 
`if (orient==LANDSCAPE)
	{
		swap(word, x1, y1);
		swap(word, x2, y2)
		y1=disp_y_size-y1;
		y2=disp_y_size-y2;
		swap(word, y1, y2)
	}`

thanks
Dario

dariocdj, 2018-02-20:
Pheraps i must set something different in setxy.h of my display ? 

`case ILI9225B:
    LCD_Write_COM_DATA(0x20,x1);
    LCD_Write_COM_DATA(0x21,y1);
    LCD_Write_COM_DATA(0x36,x2);
    LCD_Write_COM_DATA(0x37,x1);
    LCD_Write_COM_DATA(0x38,y2);
    LCD_Write_COM_DATA(0x39,y1);
    LCD_Write_COM(0x22); 
    break;`

gnulabis, 2018-02-21:
This is something that you must take upstream, with the author of UTFT. I just maintain the ESP port.

closed 2018-02-21 by gnulabis