gnulabis/UTFT-ESP #20: Is UTFT compatible with the Adafruit Huzzah + LCD Featherwing?

SoftwareGuy, 2018-03-08:
Greetings, does this library work with the Adafruit Huzzah ESP8266 and the 2.4" TFT LCD Featherwing?

Looking at the TFT Featherwing [pinouts page](https://learn.adafruit.com/adafruit-2-4-tft-touch-screen-featherwing/pinouts), it says "The TFT display, SD card and touch screen use the SPI interface to communicate. That means MISO, MOSI and SCK are used whenever either are accessed". So I assume this is SPI, not parallel mode? Digging through the documentation, it is indeed a "ILI9341" controller.

I know you may not have the device and featherwing mentioned, but if you can please provide a "looks like it'll be fine" or "might not work", I can give it a shot. Otherwise I'll stick to Adafruit's library.

gnulabis, 2018-03-08:
Hi, yes, it should work, I don't see why not.

SoftwareGuy, 2018-03-09:
Great. I'll give it a shot as soon as my new Huzzah board comes from my supplier - my current one is cooked.

SoftwareGuy, 2018-03-10:
Alright, another question - in the UTFT constructor, what is the RST argument? I have no idea what that is, all I have available to me on the TFT featherwing (see pinouts page in my original question) is:

Chip select, Data Command, Touchscreen Chip Select, SD Card Select

So putting them into the constructor I get so far:
`UTFT(ILI9341_S4P, CHIP_SELECT_PIN, ???, int SER=0);`

I would really appreciate knowing what to fill in the constructor.

gnulabis, 2018-03-10:
Actually, there is good documentation, on the original UTFT library. This repository is just a port of UTFT to ESP8266/32. 

Still, just in case, I have also included it in this repository, just look under UTFT/Documentation.

Taken from the description of the original UTFT constructor then, page 4 of the manual:

**CS:** Pin for Chip Select
**RST:** Pin for Reset
**RS:** <optional> Only used for 5pin serial modules Pin for Register Select

The third pin, RS, is always the most confusing. You can find it named as either RS, SER, or DC.

According to the datasheet and schematic of your display, there is a TFT reset pin:
https://learn.adafruit.com/adafruit-2-4-tft-touch-screen-featherwing/downloads

SoftwareGuy, 2018-03-10:
Does it absolutely require the reset pin? Or can I get away with it?
I'm not sure what GPIO controls the TFT Reset pin. I have asked around on Adafruit discord server, might try the forums next.

SoftwareGuy, 2018-03-14:
Well apparently using NOTINUSE on the UTFT constructor then using init will cause things to break when I run `UTFT::InitLCD(1)` where 0 is portrait vs 1 is landscape.

Still fruitless with the feather wing, might have to end up getting a breakout shield that supports the RST pin.
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28
Game Startup
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400d1469  PS      : 0x00060730  A0      : 0x800d0bd6  A1      : 0x3ffca610  
A2      : 0x3ffc225c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800d1430  A9      : 0x3ffca5f0  
A10     : 0x000000ff  A11     : 0x00000000  A12     : 0x3ffc1034  A13     : 0x00000003  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400e8144  LEND    : 0x400e81a6  LCOUNT  : 0x00000002  

Backtrace: 0x400d1469:0x3ffca610 0x400d0bd3:0x3ffca630 0x400e81bb:0x3ffca6b0

Rebooting...
```