gnulabis/UTFT-ESP #21: ESP32 support

jscrane, 2018-08-09:
Hi,
I've just added 8-bit support for ESP32 here: https://github.com/jscrane/UTFT-Energia

It might be good to combine our efforts? WDYT?

Cheers,
Steve

gnulabis, 2018-08-13:
Hi Steve,

In principle yes, why not? How do you see this happening, do we try to merge the two repos, and if yes, how?

Also, I tried to look at your commit that added ESP32 support, to see what you did, but it says it contains 133 (!) changed files:

https://github.com/jscrane/UTFT-Energia/commit/305059a446a9f22f30d7aed78d4191eb3ef11683

Any chance you can split the esp32 support from the rest of the updates into a separate commit?

Cheers,
Dimitris

jscrane, 2018-08-14:
Yes, you're right, my bad: I went from v2.41 of Henning Karlsen's library to v2.83 _and_ ESP32 support in one go.

As far as I can tell, all of the changes you need for 8-bit support are here: https://github.com/jscrane/UTFT-Energia/blob/master/hardware/esp32/HW_ESP32.h

(You should have all of the other infrastructure in your work already.)

I don't see how we can merge repositories easily at this point however :-(

gnulabis, 2018-08-14:
Ok, here's what I will do: I'll update first mine to v2.83 (I'm one behind, v2.82) and then I'll do a diff with yours and I'll come back to you with a proposal.