gnulabis/UTFT-ESP #25: error: unknown type name 'fontdatatype'

Dancopy, 2018-12-27:
I'm trying to compile a sketch with Nodemcu esp8266 with two additional fonts: DotMatrix_M and SevenSeg_XXXL_Num but, it is showing the following error:

Arduino: 1.8.5 (Windows 10), Card: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (1M SPIFFS)"

DotMatrix_M.c: 17: error: unknown type name 'fontdatatype'

  fontdatatype DotMatrix_M [4184] PROGMEM = {
  ^
DotMatrix_M.c: 17: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'PROGMEM'

  fontdatatype DotMatrix_M [4184] PROGMEM = {
                                 ^
exit status 1
unknown type name 'fontdatatype'

Note: The code works perfectly with the UTFT Library;
Any help, I'm grateful.

Dancopy, 2018-12-27:
First problem solved; now I need to know which pins of the Nodemcu I should connect on the TFT with a DS1307; thank you

Dancopy, 2018-12-27:
I'm making the connections as in the original UTFT library but it does not show anything on the screen.

gnulabis, 2019-01-02:
Hi and happy new year. I'm sorry I did not reply earlier, I was on holiday.

You've probably figured it out yourself, but I'm writing it here for other people who might stumble on the same issue.

When adding a .c file with a custom font definition, you need to define the "fontdatatype" type at the top of the file, just like it is done in https://github.com/gnulabis/UTFT-ESP/blob/master/UTFT/DefaultFonts.c:
```
if defined(ESP8266) || defined(ESP32)
        #include <Arduino.h>
        #include <pgmspace.h>
        #define fontdatatype const uint8_t
#endif
```

For your connection problem, I'm sorry but I cannot help you. It is impossible for me to know every possible combination of MCU and display.

closed 2019-01-02 by gnulabis

Dancopy, 2019-01-02:
Thanks and Happy New Year to you too; Yes, I have settled everything; Thank you for the excellent library.