gnulabis/UTFT-ESP #26: esp8266 Nodemcu error

Dancopy, 2019-09-05:
I don't know if this is the place indicated but I have a sketch that is showing the following error:

invalid conversion from 'const uint16_t * {aka const short unsigned int *}' to 'short unsigned int *' [-fpermissive]

Highlighted on the following line:
myGLCD.drawBitmap (x, y, 60.60, getMoonPhaseRawImage (phase));

Any tips?  Thank you

gnulabis, 2019-09-06:
Could you please provide more info?

Which version of the esp8266 platform package are you using?

Which version of Arduino IDE?

Is it possible to share your sketch?

Dancopy, 2019-09-07:
esp8266 version 2.5.2
Arduino IDE 1.8.8

`// weatherstation3
// code: https://github.com/hellange/arduino
// page: https://weatherhelge.wordpress.com/arduino/comment-page-1/?unapproved=14872&moderation-hash=e48d0dc9119c5ad6da05a5a8106c6e3e#comment-14872
//
// Weather Station with LCD TFT display
// Uses several libraries, ref link at the #include lines
// Author:
// Helge Langehaug

#include <Wire.h>
#include <UTFT.h>            // http://www.henningkarlsen.com/electronics
#include <UTouch.h>          // http://www.henningkarlsen.com/electronics
#include <Adafruit_Sensor.h>
#include <Adafruit_BMP085_U.h>
#include <Timezone.h>        // https://github.com/JChristensen/Timezone
#include <DS1307RTC.h>       // http://playground.arduino.cc//Code/Time
#include <Time.h>            // http://playground.arduino.cc//Code/Time
#include "moon_phases.h"

// Declare which fonts we will be using
extern uint8_t SmallFont[];
extern uint8_t BigFont[];
extern uint8_t SevenSegNumFont[];

// Using Arduino Mega and 3.2' TFT LCD 240x400 Display
UTFT myGLCD(TFT01_32WD,38,39,40,41);
UTouch myTouch(6,5,4,3,2);
//Adafruit_BMP085 bmp;
Adafruit_BMP085_Unified bmp = Adafruit_BMP085_Unified(10085);
boolean pressMode = false;
bool menuMode = false;

void setup(){
  Wire.begin();
  Serial.begin(9600);
  randomSeed(analogRead(0)); // ??
  
  initGraphics();
  myTouch.InitTouch();
  myTouch.setPrecision(PREC_LOW);
  //myTouch.setPrecision(PREC_MEDIUM);
  //myTouch.setPrecision(PREC_HI);
  //myTouch.setPrecision(PREC_EXTREME);

  // the function to get the time from the RTC
  setSyncProvider(RTC.get);   
  
  if(timeStatus()!= timeSet) 
     Serial.println("Unable to sync with the RTC");
  else
     Serial.println("RTC has set the system time");         
  
  // Uncomment to set date
  //setDateTime();
  

  
}

void loop(){
  myGLCD.setBackColor(0,0,0);
  renderDisplay();

  for (int i = 0; i<10; i++){
    checkTouch();
    delay(50);
  }
[... 321 lines cut ...]
  if (!bmp.begin()) {
    myGLCD.print("Could not find a valid BMP085 pressure sensor.", CENTER, 0);
    myGLCD.print("Anything connected at all ???", CENTER, 45);
  } else {
    
    int y = 150;
    myGLCD.setColor(0, 150, 0);
    myGLCD.setFont(SevenSegNumFont);
    float fTemp;
    bmp.getTemperature(&fTemp);
    //float fTemp = bmp.readTemperature();
    myGLCD.printNumI((int)fTemp, 260, y);
    float dec = (fTemp - (int)fTemp) * 10; // first decimal
    myGLCD.setFont(BigFont);
    myGLCD.printNumI(dec, 335, y+35, 1);

    myGLCD.setFont(SmallFont);
    myGLCD.print("O", 330, y-5);
    myGLCD.setFont(BigFont);
    myGLCD.print("C", 340, y);

  }
}`

Thanks

gnulabis, 2019-09-10:
This looks simply as a casting issue. The `drawbitmap()` function expects a pointer to `unsigned short int`, while the `getMoonPhaseRawImage()` apparently returns a **const** `unsigned short int`.

Just cast from one to the other or instruct your compiler to ignore this type of mismatch.

closed 2019-09-10 by gnulabis

Dancopy, 2019-09-10:
Thanks for the reply!
How would I do it? My knowledge on the subject is little!
Looks like the code was made for Mega ...

// Using Arduino Mega and 3.2 'TFT LCD 240x400 Display
UTFT myGLCD (TFT01_32WD, 38.39, 40.41);
UTouch myTouch (6,5,4,3,2);

If it really is, is it possible to use it in ESP?
Could you please take a test?
Thank you again